pauln/esphome-linp-doorbell-g03 #2: Platform should be patched via Platformio

tiberiucorbu, 2019-11-04:
The README instructs that the platform files should be patched, however PlatformIO allows to replace files from the platform with the help of build middlewares API:
https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#build-middlewares

It would be nice to handle this particular board with a build patch instead of a platform copy that can get stale.

pauln, 2019-11-06:
Hi @tiberiucorbu,
Thanks for the suggestion.  It looks as though build middlewares are a new feature in PlatformIO 4.1, which doesn't appear to have been released yet - but it's certainly good to know that it will be an option as of 4.1.

I've just had a quick look at the documentation you linked to, and it seems as though there would be a couple of options:
- The quicker/simpler option: use build middlewares to replace the static libraries with the single-core (or in this case, single-core-with-MAC-CRC-disabled) versions
- The more complete option: use build middlewares to adjust the config and rebuild the libraries on the fly

The quicker/simpler option wouldn't really address the issue of things getting stale, as the prebuilt static libraries would still get out of date and could cause strange problems if they get out of sync with the rest of the platform, so I'm not sure whether it would be worth the change.  I've no idea how to go about implementing the more complete option, though - if you've got some idea of how it would work, I'd definitely be open to a PR.

pauln, 2020-01-03:
I'm closing this issue for now.  As above, I'm definitely open to assistance on moving to build middlewares.  If @tiberiucorbu (or anyone else familiar with PlatformIO build middlewares) is willing to either contribute a PR or provide some specific guidance on how to go about this, please open a PR and/or a new issue as appropriate.

closed 2020-01-03 by pauln