pauln/esphome-linp-doorbell-g03 #4: How do I open the case?

aprosvetova, 2019-12-04:
It’s strongly pressed, I’ve tried to open the case using flat screwdriver with no success, just left some marks on the seam...

Any suggestions?

aprosvetova, 2019-12-04:
Ok, I opened it. Continued to push with a flat screwdriver, it required lots of strength, leads to lots of visible damage (eew bye-bye nice plastic) and a hole appears finally, through which you can open it completely. Happily, doorbell is not a device that hangs in your eyesight so these ‘scars’ won’t bother me.
[image]

closed 2019-12-04 by aprosvetova

pauln, 2019-12-31:
Sorry for the slow response; in case anyone else comes across this issue, I think that the case may be glued or ultrasonically welded.  I haven't found a reliable way to open it cleanly, though I did find that if you can carefully squeeze just the front plate (i.e. only the thin section in front of the seam) in a vise (so that it squeezes in from the edge), you might be able to pop it open without too much damage.  If you're careful opening it (via squeezing or prising it with a screwdriver, spudger or similar), you can press the front cover back on to the 3 pegs which hold the circuit board in, but it won't be as secure a fit (it's easy to pop the front cover off again).

I'm going to design a 3D-printed case for the doorbell which also accommodates a small circuit board with a micro-USB socket on it, which attaches to the `5V` and `GND` pads on the back of the board to power the whole unit off a USB charger.  That way, there isn't any danger of live mains voltage behind a not-very-secure front cover.

nilskt, 2020-04-23:
Please share the 3d-print files if you successfully made a case :-)

vencakratky, 2020-08-08:
Hallo, do you have somebody  3d print files for Cover? I damaged the case :-( Thank you