pauln/esphome-linp-doorbell-g03 #6: Broken OTA updates?

aprosvetova, 2019-12-05:
I use ESPHome in many of my DIY projects and love OTA updates feature.

I added `ota:` block to the doorbell config and...
It always fails on 25-35%.
It's sad because I already desoldered UART and closed the case and then wanted to update with #5 feature added.

What could be wrong with OTA updates?

pauln, 2019-12-31:
I hadn't tried an OTA update, but having just attempted it myself, I hit the exact same issue.  I watched the logs over serial while attempting an OTA, and it seems that a WDT reset is triggered - so I suspect the WDT isn't being fed during the OTA upload process.  I've added a comment with that information to an existing ESPHome issue from somebody else with the same issue: https://github.com/esphome/issues/issues/857

Snakesms, 2020-05-18:
Issue is resolved in current Dev branch, just tested and it works good.
They just added delay into the OTA update cycle:
https://github.com/esphome/esphome/pull/959

pauln, 2020-09-23:
If you're still experiencing this, please update to a recent ESPHome build.  You may need to perform one last serial flash after updating ESPHome, but then you should be able to perform OTA updates reliably.

closed 2020-09-23 by pauln