pauln/esphome-linp-doorbell-g03 #8: What pins do I need to pull to ground.

DirkTas67, 2020-03-23:
Can you explain to me what pins I need to pull to ground?
[Picture](https://photos.app.goo.gl/LvTFfd27ks6yDMQa9)

pauln, 2020-03-23:
@DirkTas67 The two pins you've marked with red are the ones which need to be pulled to ground while powering up in order to enter flashing mode.  Once you've flashed it, disconnect those pins from ground and power cycle the doorbell to perform a normal boot - it should then connect to your wifi network using the details you set in your YAML config, and you'll be able to add it to Home Assistant.

closed 2020-03-23 by pauln

fantomnotabene, 2020-04-14:
Just to clear things
[image]
[image]