pauln/esphome-linp-doorbell-g03 #12: src/linp-doorbell.h:2:19: fatal error: Queue.h: No such file or directory

DirkTas67, 2020-07-12:
In file included from src/main.cpp:50:0:
src/linp-doorbell.h:2:19: fatal error: Queue.h: No such file or directory

***************************************************************
* Looking for Queue.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Queue.h"
* Web  > https://platformio.org/lib/search?query=header:Queue.h
*
***************************************************************

compilation terminated.
*** [/data/esp32_g03_deurbel_voordeur/.pioenvs/esp32_g03_deurbel_voordeur/src/main.cpp.o] Error 1
========================= [FAILED] Took 57.66 seconds =========================

->
Looking for AsyncTCP-esphome library in registry
LibraryManager: Installing id=6798 @ 1.1.1
AsyncTCP-esphome @ 1.1.1 has been successfully installed!
Looking for ArduinoQueue library in registry
LibraryManager: Installing id=5826
ArduinoQueue @ 1.2.3 has been successfully installed!
Looking for ESPAsyncWebServer-esphome library in registry
LibraryManager: Installing id=6758 @ 1.2.6
ESPAsyncWebServer-esphome @ 1.2.6 has been successfully installed!
Looking for ESPAsyncTCP-esphome library in registry
LibraryManager: Installing id=6757
ESPAsyncTCP-esphome @ 1.2.2 has been successfully installed!
LibraryManager: Installing id=6798
Looking for ArduinoJson-esphomelib library in registry
LibraryManager: Installing id=3837 @ 5.13.3
ArduinoJson-esphomelib @ 5.13.3 has been successfully installed!

->
Any ideas's?

Regards
Dirk

vencakratky, 2020-07-14:
I have the same Issue on compiling on Home Assistant ESPHome 
`In file included from src/main.cpp:32:0:
src/linp-doorbell.h:2:19: fatal error: Queue.h: No such file or directory

***************************************************************
* Looking for Queue.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Queue.h"
* Web  > https://platformio.org/lib/search?query=header:Queue.h
*
***************************************************************

compilation terminated.
*** [/data/linp_doorbell/.pioenvs/linp_doorbell/src/main.cpp.o] Error 1
========================= [FAILED] Took 49.08 seconds =========================`

Thomasv1212, 2020-07-15:
Anyone a solution jet?

bonuzzz, 2020-07-16:
solution is here:
https://github.com/pauln/esphome-linp-doorbell-g03/pull/11

DirkTas67, 2020-07-16:
if you look in my first message you see the following -> Looking for ArduinoQueue library in registry

So the library is loaded, I downloaded the latest version of the files here but still get this error.

bonuzzz, 2020-07-16:
but also you should edit linp-doorbell.h file
https://github.com/pauln/esphome-linp-doorbell-g03/pull/11/files

vencakratky, 2020-07-17:
Yes, after edit linp-doorbell.h it´s working. Thank you :) 
[image]

DirkTas67, 2020-07-19:
Thank you, this is solved.

closed 2020-07-19 by DirkTas67