pauln/esphome-linp-doorbell-g03 #13: Chime is being heard only after 2nd push of the button

lustyffh, 2020-09-22:
Hi, I have an integration of this doorbell with HA. Such as when binary_sensor.button_1 is turned on I get a Telegram notification. What I observe now is that I always get Telegram notify when remote button is pushed first time. However, the doorbell executes chime only after 2nd push of the button (and 2nd Tlgrm notification)

vencakratky, 2020-09-23:
This is true. I remember, with original firmware, after Power on played tunes. With this firmware nothing. I have to press the button every time I turn on the Doorbell

pauln, 2020-09-23:
That's interesting - I've also noticed that sometimes the chime doesn't play when the button is pressed - but my HA automation also fires on the silent press, and the next press triggers the chime (and HA automation/notification).  I'll have to set up a unit for development and have another poke at it - from memory, the chime on press is handled by the secondary microcontroller, so I'm not sure why it sometimes fails to do its job, since we're not touching its firmware.

closed 2020-09-23 by pauln

pauln, 2020-09-23:
It turns out that the secondary microcontroller (responsible for the actual doorbell functionality) remembers the volume level across restarts - but for some reason, it doesn't actually apply it on boot.  The stock firmware seems to work around this by setting the volume as part of its boot sequence (as noted by @vencakratky, this causes the chime to play when the doorbell powers up).

I've just tagged v0.5 which includes a fix for this - but note that since the fix is to mimic the stock firmware and set the volume on boot, this means that *the chime will now be played on boot*.  This also fixes the (unreported) issue where the volume was resetting to the default on boot, which may not be the volume you'd set.

commit 92d7df09 (2020-09-23): Reset volume on boot

Although the secondary microcontroller remembers the volume, it doesn't apply it on boot.
We therefore need to set it once we receive the response to the get_volume call we make on boot in order to avoid it defaulting to a setting which may not be what the user wants.
This also fixes the "first button press doesn't trigger chime" issue / closes #13.