pauln/esphome-linp-doorbell-g03 #17: Help, identify resistor

stevoh6, 2021-04-29:
Hi, please could somebody make picture of his board, or write here params of this resistor? He didnt make it. 
Thanks.
[image]

pauln, 2021-04-30:
Wow, how did you manage that?!  It appears to be a 47 ohm resistor (5% tolerance) - the colours are yellow/violet/black/gold.

closed 2021-04-30 by pauln

stevoh6, 2021-04-30:
Thank you :-) I pug it into socket and it make a puff :D. Anyway, how many watts it should handle? 1W, 2w or 3w?

pauln, 2021-04-30:
Based on the size, I'd guess 1W.  Hopefully it did its job as a fuse successfully and protected the rest of the board - and hopefully you can figure out what happened and stop it from happening again!

stevoh6, 2021-04-30:
I will let you know, I'm guessing two things: 
1) I little adjust position of this resistor to be able put it into the plastic cover (there was a crack after puff, maybe this crack was here before)
2) not properly clean board from flux after soldering