pauln/esphome-linp-doorbell-g03 #18: Flash issues

numminorih-git, 2021-04-29:
First of all thank you for your project!

Can you please help me with flashing doorbell. All connected as per manual (photo attached), but I still can't flash it (after plugin serial-to-USB adapter to usb I listen standart doorbell boot "dingding" sound):

#
Use manually specified: /dev/ttyUSB0
Uploading .pioenvs/linp_doorbell/firmware.bin
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____....._____....._____....._____
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
*** [upload] Error 2
#

[image]

pauln, 2021-04-30:
@numminorih-git I can't see which pins on your serial-to-USB adapter you've got the wires connected to, but if you're sure that your serial-to-USB adapter works and that it's attaching to `/dev/ttyUSB0`, make sure you have it wired as:
- GND to GND
- 3.3v to 3.3v (don't connect it to 5v - if your serial adapter doesn't have a 3.3v pin or can't provide enough current, power the doorbell via the 5V and GND pads on the other side of the board)
- TXD0 to RX (doorbell transmits, serial adapter receives)
- RXD0 to TX (doorbell receives, serial adapter transmits)

It also looks like you've stuffed the ends of a couple of the wires into female plugs, which might not be making good connections.  If you can spare the wires, it might be better to cut the plugs off that end of the black and white wires, and then solder the black/white/purple/red wires onto the doorbell board to ensure you're getting a good connection.

The ESP32 might take more current to actually write to the flash than your serial-to-USB adapter can provide, so if you have a 5V power supply you can attach to the pads on the other side of the board (even an old USB cable that you can cut up - but make sure to check the polarity of the wires with a multimeter first; not all USB cables use normal colour conventions!), that might be a good option.  If you do power it separately, make sure that both the power supply and the serial-to-USB adapter are connected to ground (GND) pads on the doorbell, so that they share a common ground.  With my serial adapter, I find that I have to plug it in first (before the power), or it won't work properly (I guess it gets into a weird state if it gets any power backfed on the serial lines before it's been powered up properly), but yours might vary.

You might also want to only solder the "flashing mode" wires to the corner pads, not to ground - just touch them to that ground pad while powering it up to get into flashing mode.  Once you've booted it into flashing mode, you can disconnect them from ground (you'll need to disconnect them from ground to boot into "normal" mode once the flashing is complete).

numminorih-git, 2021-04-30:
@pauln thank you for your prompt reply

GND to GND
3.3v to empty
TXD0 to RX 
RXD0 to TX 

on the back of doorbell
+5 v and GND to external power supply (5А+)
add. GND connected to GND of USB stick (I try use stick and 2 different arduino on different chipset)

1. usb stick/arduino connected to PC
2. "2 corner wires" connected to general GND
3. Power up external power
4. Doorbell is booting up and sounds standard ringtone (it sound in this case whatever "2 corner wires" connected to general GND or/ inked together or/ hang in the air or/ hang in the air after boot)
5.  usb stick/arduino Rx start blinking blue - as far as I understand - sending packets
6. error when trying to flash

I'm stuck...

Should the bell sounded at boot with "2 corner wires" connected to general GND as without it?

pauln, 2021-05-01:
It's been quite a while since I cracked one of these open, and I almost always have the speaker unplugged when I'm working on them (because it's easier to work with the PCB when it's not attached to the plastic case), so I don't remember whether the chime sounds when you successfully get it into flash mode for the first time, sorry.  I've got one here (which I flashed with ESPHome long ago) which I use for development, and the blue LEDs do light up for a second or two, whether booting into flash mode or normal mode - but there's no noise.  It's possible that they've made a change (possibly to the secondary microcontroller's firmware) since I bought my doorbells, which triggers an audible chime on boot - in which case it might be normal for it to happen when entering flash mode as well as on a normal boot.

I've used a few different serial adapters, but the ones I mostly use these days are based on the CP2102.  It might be worth using a serial logger/monitor application to check whether you're receiving anything from the ESP32 when it boots (I think it was pretty standard 115200 8N1) - if it's in flash mode, you should see it spit out something along these lines:
```
rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
```
If you see a whole bunch of other logs go by (a lot of which mention "miio"), it's in normal mode.  If you're able to see logs this way, and it's getting into flash mode but you still can't flash it, double check the USB-serial TX to ESP32 RXD0 connection to make sure it's a good connection and not shorted to ground (or anything else) - it might be that the ESP32 is in flash mode but just isn't receiving the commands from your computer.

numminorih-git, 2021-05-03:
Ouh, thank you for your support. I bought USB-TTL CP2102 Robotdyn and know have some add. info:

1. 
```
Serial port /dev/ttyUSB0
Connecting....
Chip is unknown ESP32 (revision 1)
Features: WiFi, BT, Single Core, 240MHz, VRef calibration in efuse, Coding Scheme 3/4
MAC: 64:90:c1:98:ec:2c
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0x0, SizeID=0x0), defaulting to 4MB
Compressed 15872 bytes to 10319...

Writing at 0x00001000... (100 %)
Wrote 15872 bytes (10319 compressed) at 0x00001000 in 0.9 seconds (effective 138.9 kbit/s)...

A fatal error occurred: Timed out waiting for packet header
*** [upload] Error 2
```

2. 
```
chip@io esptool.py --port /dev/ttyUSB0 read_mac                                          
esptool.py v2.8
Serial port /dev/ttyUSB0
Connecting........
Detecting chip type... ESP32
Chip is unknown ESP32 (revision 1)
Features: WiFi, BT, Single Core, 240MHz, VRef calibration in efuse, Coding Scheme 3/4
Crystal is 40MHz
MAC: 64:90:c1:98:ec:2c
Uploading stub...
Running stub...
Stub running...
MAC: 64:90:c1:98:ec:2c
Hard resetting via RTS pin...
```

Some interesting info
I can read mac (2) but only at first time after powerup external power supply, if i try to read it again I'l see "Connecting........_____....." until restart external power

closed 2021-05-09 by pauln