pauln/esphome-linp-doorbell-g03 #21: Can't compile

000al000, 2021-11-27:
PlatformIO Core, version 5.2.2
Esphome Version: 2021.11.3

Debian Python 3.7

```
esphome:
  name: doorbell
  platform: ESP32
  board: esp32doit-devkit-v1
  platformio_options:
    platform: espressif32@1.11.0
    platform_packages: |-4

          framework-arduinoespressif32 @ https://github.com/pauln/arduino-esp32.git#solo-no-mac-crc/1.0.4

  includes:
    - linp-doorbell.h
  libraries:
    - "ArduinoQueue"

wifi:
  ssid: 'your-wifi-ssid-here'
  password: 'put-your-wifi-password-here'

```

Can't complie, got error

INFO Reading configuration doorbell.yaml...
INFO Generating C++ source...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/dist-packages/esphome/__main__.py", line 817, in main
    return run_esphome(sys.argv)
  File "/usr/local/lib/python3.7/dist-packages/esphome/__main__.py", line 804, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/usr/local/lib/python3.7/dist-packages/esphome/__main__.py", line 324, in command_compile
    exit_code = write_cpp(config)
  File "/usr/local/lib/python3.7/dist-packages/esphome/__main__.py", line 156, in write_cpp
    generate_cpp_contents(config)
  File "/usr/local/lib/python3.7/dist-packages/esphome/__main__.py", line 168, in generate_cpp_contents
    CORE.flush_tasks()
  File "/usr/local/lib/python3.7/dist-packages/esphome/core/__init__.py", line 613, in flush_tasks
    self.event_loop.flush_tasks()
  File "/usr/local/lib/python3.7/dist-packages/esphome/coroutine.py", line 245, in flush_tasks
    next(task.iterator)
  File "/usr/local/lib/python3.7/dist-packages/esphome/core/config.py", line 283, in _add_platformio_options
    cg.add_platformio_option(key, val)
  File "/usr/local/lib/python3.7/dist-packages/esphome/cpp_generator.py", line 595, in add_platformio_option
    CORE.add_platformio_option(key, value)
  File "/usr/local/lib/python3.7/dist-packages/esphome/core/__init__.py", line 707, in add_platformio_option
    assert isinstance(value, list)
AssertionError

000al000, 2021-11-30:
try, but got 
src/esphome/components/esp32/core.cpp:8:29: fatal error: esp_idf_version.h: No such file or directory

GenoMXXX, 2021-11-30:
I solved this problem using ESPHome Version 1.20.0, but I can't connect to ESP with FlashDownloaderTool

pauln, 2021-11-30:
It looks as though the changes introduced in ESPHome 2021.10 to allow the choice of framework (ESP-IDF or Arduino) have broken this config.  It should hopefully mean that a "nicer" config is possible (without having to use the weird whitespace for the platformio options), but it looks as though there may be other changes required to support 2021.10 and above.  I'll look into that in the near future, but until then, you can roll back to ESPHome 2021.9 in order to use this (either the old custom component or the newer external component version as suggested by @bonuzzz - that's also my recommendation; I'll probably swap things around to make the main branch of this repo the external component, so that it's the first version that new users see).

000al000, 2021-11-30:
thank you for answers

As i see i can compile without external_components (original version) 

i modify original yaml to

esphome:
  name: doorbell

  platformio_options:
    platform: espressif32@3.2.0
    platform_packages:
      - framework-arduinoespressif32 @ https://github.com/pauln/arduino-esp32.git#solo-no-mac-crc/1.0.6
  platform: ESP32
  board: esp32doit-devkit-v1

and it compile without error.

Can i use this firmware to flash to ESP?

pauln, 2021-11-30:
@000al000 Ah, that's a good catch - it looks as though the `platformio_options` config now supports a YAML list for `platform_packages`, so if you're lucky, that change might indeed to the trick.  If you're getting a clean build, yes, you should be ok to flash it to the ESP - but I'd still recommend using the `external_components` version (just make the same change to the `platform_packages` config so that it builds).

000al000, 2021-11-30:
external_components had another problem (in oroginal version i fixed it to add this function to linp-doorbell.h from old esphome version )

```
Compiling .pioenvs/linp_doorbell/src/esphome/components/ota/ota_component.cpp.o
Compiling .pioenvs/linp_doorbell/src/esphome/components/sensor/automation.cpp.o
Compiling .pioenvs/linp_doorbell/src/esphome/components/sensor/filter.cpp.o
src/esphome/components/linp_doorbell/linp_doorbell.cpp: In member function 'void esphome::linp_doorbell::LinpDoorbellComponent::handleEvent(String)':
src/esphome/components/linp_doorbell/linp_doorbell.cpp:129:75: error: 'parse_float' was not declared in this scope
       this->chime_playing_sensor_->publish_state(parse_float(event.c_str()).value());
                                                                           ^
src/esphome/components/linp_doorbell/linp_doorbell.cpp: In member function 'void esphome::linp_doorbell::LinpDoorbellComponent::handleParam(String, String)':
src/esphome/components/linp_doorbell/linp_doorbell.cpp:155:68: error: 'parse_float' was not declared in this scope
       this->volume_sensor_->publish_state(parse_float(value.c_str()).value());
                                                                    ^
src/esphome/components/linp_doorbell/linp_doorbell.cpp:166:48: error: 'parse_float' was not declared in this scope
       auto tuneFloat = parse_float(tune.c_str());
                                                ^
*** [.pioenvs/linp_doorbell/src/esphome/components/linp_doorbell/linp_doorbell.cpp.o] Error 1
======================================================================================================== [FAILED] Took 22.72 seconds =
```

000al000, 2021-12-01:
i compile  with external_componets with esphome 2021.9 with standart yaml
only replace

```
    platform_packages:.
       - framework-arduinoespressif32 @ https://github.com/pauln/arduino-esp32.git#solo-no-mac-crc/1.0.**4**
   to

    platform_packages:.
       - framework-arduinoespressif32 @ https://github.com/pauln/arduino-esp32.git#solo-no-mac-crc/1.0.**6**

```

flash it success, but nothing work (no reaction on button, no play tune if call service from ha)
in log i see CRC ERROR

my doorbell model **G6L-Wifi**

```
INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from doorbell.local using esphome API
INFO Successfully connected to doorbell.local
[20:30:18][I][app:102]: ESPHome version 2021.9.0 compiled on Dec  1 2021, 20:29:44
[20:30:18][C][wifi:501]: WiFi:
[20:30:18][C][wifi:361]:   SSID: 'PASTE$COPY'
[20:30:18][C][wifi:362]:   IP Address: 10.100.10.94
[20:30:18][C][wifi:364]:   BSSID: 34:CE:00:65:30:79
[20:30:18][C][wifi:365]:   Hostname: 'doorbell'
[20:30:18][C][wifi:369]:   Signal strength: -56 dB ▂▄▆█
[20:30:18][C][wifi:373]:   Channel: 1
[20:30:18][C][wifi:374]:   Subnet: 255.255.255.0
[20:30:20][D][api:095]: Accepted 10.100.10.200
[20:30:20][D][api.connection:735]: Home Assistant 2021.11.4 (10.100.10.200): Connected successfully
[20:30:20][D][esp-idf:000]: E (6448) system_api: Base MAC address from BLK0 of EFUSE CRC error, efuse_crc = 0x9a; calc_crc = 0x31
```

bonuzzz, 2021-12-01:
esphome:
  name: $devicename
  platform: ESP32
  board: esp32doit-devkit-v1
  build_path: build/$devicename  

  platformio_options:
    platform: espressif32@1.11.0
    platform_packages: |-4
          framework-arduinoespressif32 @ https://github.com/pauln/arduino-esp32.git#solo-no-mac-crc/1.0.4
 

Compiled on 2021.8

000al000, 2021-12-02:
@bonuzzz - thank you

closed 2021-12-02 by 000al000

DirkTas67, 2022-05-26:
Hello,

I have the same problem, I never installed new firmware since version 1.20.0 Jul 22 2021, 14:42:04 because of this problem, my version of ESPHome is 2022.2.4. And now I have changed my Network, at least 95% of the devices I have have changed network but my 2 linp devices I can't changed, I still have the original network so I still can access them but now I can't compile the new firmware. 

somebody can help me with this?

substitutions:
  display_name: linptech1
  device_name: linptech01
  device_description: ESP32 Linp G03 Deurbel
  friendly_name: ESP32 Linp Deurbel 1
  
esphome:
  name: ${display_name}
  platform: ESP32
  board: esp32doit-devkit-v1
  includes:
    - linp-doorbell.h
  libraries:
    - "ArduinoQueue"
  platformio_options:
    platform: espressif32@1.11.0
    platform_packages:
    - framework-arduinoespressif32 @ https://github.com/pauln/arduino-esp32.git#solo-no-mac-crc/1.0.6

bonuzzz, 2022-05-26:
@DirkTas67 see here https://github.com/vampywiz17/esphome-linp-doorbell-g03/commits/feature/external_components

DirkTas67, 2022-05-26:
@bonuzzz 

This is what you problably tried to tell me

my custom_components/linp_doorbell/sensor.py now looks like this

import esphome.codegen as cg
import esphome.config_validation as cv
from esphome.components import sensor
from esphome.const import CONF_ID, UNIT_EMPTY

linp_doorbell_ns = cg.esphome_ns.namespace("linp_doorbell")

LinpDoorbellComponent = linp_doorbell_ns.class_(
    "LinpDoorbellComponent", cg.Component
)

CONF_VOLUME = "volume"
CONF_CHIME_PLAYING = "chime_playing"
CONF_USE_OLD_SERVICE_NAMES = "use_old_service_names"

CONFIG_SCHEMA = (
    cv.Schema(
        {
            cv.GenerateID(): cv.declare_id(LinpDoorbellComponent),
            cv.Optional(CONF_VOLUME): sensor.sensor_schema(
                unit_of_measurement=UNIT_EMPTY,
                accuracy_decimals=0,
            ),
            cv.Optional(CONF_CHIME_PLAYING): sensor.sensor_schema(
                unit_of_measurement=UNIT_EMPTY,
                accuracy_decimals=0,
            ),
            cv.Optional(CONF_USE_OLD_SERVICE_NAMES, default=False): cv.boolean,
        }
    )
)

bonuzzz, 2022-05-26:
Ok. Just copy content from my link above as usual custom component in esphome to custom_components folder. It was breaking change in begin of year. So simple solution is to use component from that repository.

```
  platform: ESP32
  board: esp32doit-devkit-v1
  platformio_options:
    platform: espressif32@3.2.0
    platform_packages:
       - framework-arduinoespressif32 @ https://github.com/pauln/arduino-esp32.git#solo-no-mac-crc/1.0.6
```

DirkTas67, 2022-05-26:
@bonuzzz 

I want to thank you for your help but it doesn't seem to help, first a question
does the below lines have to be included? Right now they are gone.

  includes:
    - linp-doorbell.h
  libraries:
    - "ArduinoQueue"

the result I get is this

Compiling .pioenvs/linptech1/src/main.cpp.o
Compiling .pioenvs/linptech1/FrameworkArduino/wiring_shift.c.o
Archiving .pioenvs/linptech1/libFrameworkArduino.a
/config/g03_deurbel.yaml: In lambda function:
/config/g03_deurbel.yaml:44:27: error: expected type-specifier before 'LinpDoorbell'
     auto doorbell = new LinpDoorbell();
                           ^
/config/g03_deurbel.yaml:46:23: error: could not convert '{doorbell}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::Component*>'
/config/g03_deurbel.yaml: In lambda function:
/config/g03_deurbel.yaml:60:41: error: 'get_linp_doorbell' was not declared in this scope
     lambda: return {get_linp_doorbell(doorbell)->volume_sensor};
                                         ^
/config/g03_deurbel.yaml:60:57: error: could not convert '{<expression error>}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::sensor::Sensor*>'
     lambda: return {get_linp_doorbell(doorbell)->volume_sensor};
                                                         ^
/config/g03_deurbel.yaml: In lambda function:
/config/g03_deurbel.yaml:65:41: error: 'get_linp_doorbell' was not declared in this scope
     lambda: return {get_linp_doorbell(doorbell)->playing_sensor};
                                         ^
/config/g03_deurbel.yaml:65:58: error: could not convert '{<expression error>}' from '<brace-enclosed initializer list>' to 'std::vector<esphome::sensor::Sensor*>'
     lambda: return {get_linp_doorbell(doorbell)->playing_sensor};
                                                          ^
/config/g03_deurbel.yaml: In lambda function:
/config/g03_deurbel.yaml:71:61: error: 'get_linp_doorbell' was not declared in this scope
       return std::vector<Sensor*>(get_linp_doorbell(doorbell)->chime_sensors, std::end(get_linp_doorbell(doorbell)->chime_sensors));
                                                             ^
/config/g03_deurbel.yaml: In lambda function:
/config/g03_deurbel.yaml:90:67: error: 'get_linp_doorbell' was not declared in this scope
       return std::vector<BinarySensor*>(get_linp_doorbell(doorbell)->button_sensors, std::end(get_linp_doorbell(doorbell)->button_sensors));
                                                                   ^
*** [.pioenvs/linptech1/src/main.cpp.o] Error 1
========================== [FAILED] Took 9.34 seconds ==========================

bonuzzz, 2022-05-27:
@DirkTas67 this is my config

```
substitutions:
  devicename: doorbell
  
esphome:
  name: $devicename
  build_path: build/$devicename
  platform: ESP32
  board: esp32doit-devkit-v1
  platformio_options:
    platform: espressif32@3.2.0
    platform_packages:
       - framework-arduinoespressif32 @ https://github.com/pauln/arduino-esp32.git#solo-no-mac-crc/1.0.6

ota:

api:
  reboot_timeout: 15min
  
logger:
  level: NONE

wifi:
  ssid: 'ssid'
  password: 'pass'

sensor:
  - platform: linp_doorbell
    volume:
      name: "Doorbell volume"
    chime_playing:
      name: "Doorbell playing tune"
```

DirkTas67, 2022-06-03:
@bonuzzz 

I took your yaml file and replace mine, cleaned the build file, placed the linp_doorbell folder you attached and put it in the custom_components folder and hit install.... This is what I get, there must be something wrong with my esphome docker container, I'm in the process to replace the docker container (and a lot of other docker containers) to Truenas Scale, I have set esphome up and it works but there is a 'small' problem, I can't find the config files even after I have defined the /config folder to the one I want it to use.

another thing: in the past I had this in my yaml, is this gone now?

  - platform: custom
    lambda: return {get_linp_doorbell(doorbell)->volume_sensor};
    sensors:
      - name: ${friendly_name} Volume
  # Doorbell "chime playing" sensor (optional)
  - platform: custom
    lambda: return {get_linp_doorbell(doorbell)->playing_sensor};
    sensors:
      - name: ${friendly_name} Tune Playing
  # Tune/Chime configuration (one per button, optional)
  - platform: custom
    lambda: |-
      return std::vector<Sensor*>(get_linp_doorbell(doorbell)->chime_sensors, std::end(get_linp_doorbell(doorbell)->chime_sensors));
    sensors:
      - name: ${friendly_name} Button 1 Tune
      - name: ${friendly_name} Button 2 Tune
      - name: ${friendly_name} Button 3 Tune
      - name: ${friendly_name} Button 4 Tune
      - name: ${friendly_name} Button 5 Tune
      - name: ${friendly_name} Button 6 Tune
      - name: ${friendly_name} Button 7 Tune
      - name: ${friendly_name} Button 8 Tune
      - name: ${friendly_name} Button 9 Tune
      - name: ${friendly_name} Button 10 Tune

# Binary sensors for button presses (optional)
binary_sensor:
  - platform: status
    name: ${display_name} Status
  - platform: custom
    lambda: |-
      return std::vector<BinarySensor*>(get_linp_doorbell(doorbell)->button_sensors, std::end(get_linp_doorbell(doorbell)->button_sensors));
    binary_sensors:
      - name: ${friendly_name} Button 1
      - name: ${friendly_name} Button 2
      - name: ${friendly_name} Button 3
      - name: ${friendly_name} Button 4
      - name: ${friendly_name} Button 5
      - name: ${friendly_name} Button 6
      - name: ${friendly_name} Button 7
      - name: ${friendly_name} Button 8
      - name: ${friendly_name} Button 9
      - name: ${friendly_name} Button 10

Is it possible that you make this bin file for me, as wifi credentials it doesn't really matter as long as the AP is set to something we know,  I don't see another possibility right now, unless you know something better.

INFO Reading configuration /config/linptech01.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing doorbell (board: esp32doit-devkit-v1; framework: arduino; platform: espressif32@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-arduinoespressif32 0.0.0+sha.a205eef 
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Library Manager: Installing ArduinoQueue @ 1.2.3

Unpacking  [------------------------------------]    0%
Unpacking  [###---------------------------------]    8%
Unpacking  [######------------------------------]   16%
Unpacking  [#########---------------------------]   25%
Unpacking  [############------------------------]   33%
Unpacking  [###############---------------------]   41%
Unpacking  [##################------------------]   50%
Unpacking  [#####################---------------]   58%
[... 21 lines cut ...]
Compiling .pioenvs/doorbell/src/esphome/components/esp32/core.cpp.o
Compiling .pioenvs/doorbell/src/esphome/components/esp32/gpio_arduino.cpp.o
Compiling .pioenvs/doorbell/src/esphome/components/esp32/gpio_idf.cpp.o
Compiling .pioenvs/doorbell/src/esphome/components/esp32/preferences.cpp.o
Compiling .pioenvs/doorbell/src/esphome/components/linp_doorbell/linp_doorbell.cpp.o
Compiling .pioenvs/doorbell/src/esphome/components/logger/logger.cpp.o
Compiling .pioenvs/doorbell/src/esphome/components/md5/md5.cpp.o
Compiling .pioenvs/doorbell/src/esphome/components/mdns/mdns_component.cpp.o
Compiling .pioenvs/doorbell/src/esphome/components/mdns/mdns_esp32_arduino.cpp.o
src/esphome/components/linp_doorbell/linp_doorbell.cpp: In member function 'void esphome::linp_doorbell::LinpDoorbellComponent::handleEvent(String)':
src/esphome/components/linp_doorbell/linp_doorbell.cpp:129:75: error: 'parse_float' was not declared in this scope
       this->chime_playing_sensor_->publish_state(parse_float(event.c_str()).value());
                                                                           ^
src/esphome/components/linp_doorbell/linp_doorbell.cpp: In member function 'void esphome::linp_doorbell::LinpDoorbellComponent::handleParam(String, String)':
src/esphome/components/linp_doorbell/linp_doorbell.cpp:155:68: error: 'parse_float' was not declared in this scope
       this->volume_sensor_->publish_state(parse_float(value.c_str()).value());
                                                                    ^
src/esphome/components/linp_doorbell/linp_doorbell.cpp:166:48: error: 'parse_float' was not declared in this scope
       auto tuneFloat = parse_float(tune.c_str());
                                                ^
Compiling .pioenvs/doorbell/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling .pioenvs/doorbell/src/esphome/components/mdns/mdns_esp_idf.cpp.o
Compiling .pioenvs/doorbell/src/esphome/components/network/util.cpp.o
*** [.pioenvs/doorbell/src/esphome/components/linp_doorbell/linp_doorbell.cpp.o] Error 1
========================= [FAILED] Took 28.77 seconds =========================

DirkTas67, 2022-06-03:
[image]

DirkTas67, 2022-06-03:
YES!

ScorpionSPB, 2022-11-05:
Hello, can you help? I'm stuck at the compilation step too.
I downloaded the example posted above.
and I get a compilation error.
My config:
`esphome:
  name: linp_doorbell
  platform: ESP32
  board: esp32doit-devkit-v1
  platformio_options:
    platform: espressif32@1.11.0
    platform_packages: |-4
      
          framework-arduinoespressif32 @ https://github.com/pauln/arduino-esp32.git#solo-no-mac-crc/1.0.4
          
external_components:
  - source: github://pauln/esphome-linp-doorbell-g03@feature/external_components
    components: [ linp_doorbell ]

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# Enable Home Assistant API
api:
  password: !secret api_password

# Enable over-the-air updates
ota:
  password: !secret ota_password

sensor:
  - platform: linp_doorbell
    volume:
      name: "Doorbell volume"
    chime_playing:
      name: "Doorbell playing tune"
    use_old_service_names: True`

The error I am getting
`INFO Reading configuration /root/config/linp_doorbell.yaml...
ERROR Unable to load component linp_doorbell.sensor:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/root/config/.esphome/external_components/10edc280/components/linp_doorbell/sensor.py", line 20, in <module>
    cv.Optional(CONF_VOLUME): sensor.sensor_schema(
TypeError: sensor_schema() takes from 0 to 1 positional arguments but 3 were given
WARNING 'linp_doorbell': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
Failed config

sensor.linp_doorbell: [source /root/config/linp_doorbell.yaml:28]
  
  Platform not found: 'sensor.linp_doorbell'.
  platform: linp_doorbell
  volume: 
    name: Doorbell volume
  chime_playing: 
    name: Doorbell playing tune
  use_old_service_names: True`

bonuzzz, 2022-11-05:
@ScorpionSPB I don't use external_components. Try to manually place module code to custom_components folder

ScorpionSPB, 2022-11-06:
Could you please re-upload the folder
__init__.py file - should really be 0 kb
i get this error:
```
INFO Reading configuration /root/config/linp-doorbell.yaml...
Failed config

sensor.linp_doorbell: [source /root/config/linp-doorbell.yaml:27]
  
  Platform not found: 'sensor.linp_doorbell'.
  platform: linp_doorbell
  volume: 
    name: Doorbell volume
  chime_playing: 
    name: Doorbell playing tune
```

bonuzzz, 2022-11-06:
I re-checked. Everything is ok in that archive. 
[image]

Check the path where you place this component. It should be in custom_components folder. In you case /root/config/custom_components/linp_doorbell
[image]

ScorpionSPB, 2022-11-06:
There is some success
I'm in this place right now:

```
INFO Reading configuration /root/config/linp-doorbell.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing doorbell (board: esp32doit-devkit-v1; framework: arduino; platform: espressif32@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-arduinoespressif32 @ 0.0.0+sha.a205eef 
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- WiFi @ 1.0
|-- ESPmDNS @ 1.0
|-- Update @ 1.0
|-- ArduinoQueue @ 1.2.3
Compiling .pioenvs/doorbell/src/main.cpp.o
Linking .pioenvs/doorbell/firmware.elf
RAM:   [=         ]  11.7% (used 38276 bytes from 327680 bytes)
Flash: [====      ]  44.6% (used 817958 bytes from 1835008 bytes)
Building .pioenvs/doorbell/firmware.bin
esp32_create_combined_bin([".pioenvs/doorbell/firmware.bin"], [".pioenvs/doorbell/firmware.elf"])
Wrote 0xd7ba0 bytes to file /root/config/build/doorbell/.pioenvs/doorbell/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 8.38 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.3.6
ERROR Connecting to 192.168.3.6:3232 failed: [Errno 111] Connection refused
```

I registered, changed the static IP and the config to:

```
substitutions:
  devicename: doorbell
  
esphome:
  name: $devicename
  build_path: build/$devicename
  platform: ESP32
  board: esp32doit-devkit-v1
  platformio_options:
    platform: espressif32@3.2.0
    platform_packages:
       - framework-arduinoespressif32 @ https://github.com/pauln/arduino-esp32.git#solo-no-mac-crc/1.0.6

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  manual_ip:
    static_ip: 192.168.3.6
    gateway: 192.168.3.1
    subnet: 255.255.255.0

# Enable Home Assistant API
api:
  password: !secret api_password

# Enable over-the-air updates
ota:
  password: !secret ota_password

sensor:
  - platform: linp_doorbell
    volume:
      name: "Doorbell volume"
    chime_playing:
      name: "Doorbell playing tune"
```

bonuzzz, 2022-11-06:
If you flash esphome firmware to this doorbell for the first time, you should do it manually following the instruction on this readme https://github.com/pauln/esphome-linp-doorbell-g03/blob/master/README.md

ScorpionSPB, 2022-11-06:
No no. The patch is already installed. And works. My task is to move gateway from 192.168.3.1 to 192.168.1.1
To do this, I need to rewrite the config.

bonuzzz, 2022-11-06:
I think it's better to use dhcp here

pauln, 2022-11-07:
@ScorpionSPB I've just pushed some updates which should fix compilation issues on the `external_components` branch - so if you update your config in line with the changes to the sample yaml file and try again, you should hopefully get a clean/working build now.  If you've recently tried using the external component, you may have the previous version cached in which case you may need to temporarily add `refresh: 0s` to the `external_components` block for this repo in order to force it to fetch a fresh copy.

If you're trying to change the wifi config via OTA, you might need to use `use_address: 192.168.3.6` temporarily to force ESPHome to use the old IP address to connect while applying the new config.  See the [ESPHome docs on changing a node's name](https://esphome.io/components/esphome.html#esphome-changing-node-name), as it's a similar concept.

pauln, 2022-11-08:
It's on the [feature/external-components branch](https://github.com/pauln/esphome-linp-doorbell-g03/tree/feature/external_components).

ScorpionSPB, 2022-11-08:
It looks like you still have to pick up a soldering iron (soldering dryer)...
```
INFO Reading configuration /config/esphome/linp_doorbell.yaml...
WARNING 'linp_doorbell': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Processing linp_doorbell (board: esp32doit-devkit-v1; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
Platform Manager: Installing platformio/espressif32 @ 3.5.0
INFO Installing platformio/espressif32 @ 3.5.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif32@3.5.0 has been installed!
INFO espressif32@3.5.0 has been installed!
Tool Manager: Installing platformio/toolchain-xtensa32 @ ~2.50200.0
INFO Installing platformio/toolchain-xtensa32 @ ~2.50200.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%
Tool Manager: toolchain-xtensa32@2.50200.97 has been installed!
INFO toolchain-xtensa32@2.50200.97 has been installed!
Tool Manager: Installing git+https://github.com/pauln/arduino-esp32.git#solo-no-mac-crc/1.0.6
INFO Installing git+https://github.com/pauln/arduino-esp32.git#solo-no-mac-crc/1.0.6
git version 2.30.2
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-407yc7xr'...
Submodule 'libraries/AzureIoT' (https://github.com/VSChina/ESP32_AzureIoT_Arduino) registered for path 'libraries/AzureIoT'
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-407yc7xr/libraries/AzureIoT'...
Submodule path 'libraries/AzureIoT': checked out '5e8ffb21115675f8c258e81bb287a9cd610e3205'
Tool Manager: framework-arduinoespressif32@0.0.0+sha.a205eef has been installed!
INFO framework-arduinoespressif32@0.0.0+sha.a205eef has been installed!
Tool Manager: Installing platformio/tool-esptoolpy @ ~1.30100.0
INFO Installing platformio/tool-esptoolpy @ ~1.30100.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-esptoolpy@1.30100.210531 has been installed!
INFO tool-esptoolpy@1.30100.210531 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40300.0
INFO Installing platformio/tool-scons @ ~4.40300.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-scons@4.40300.220725 has been installed!
INFO tool-scons@4.40300.220725 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-arduinoespressif32 @ 0.0.0+sha.a205eef 
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Library Manager: Installing ArduinoQueue @ 1.2.3

Downloading  [------------------------------------]    0%
Downloading  [####################################]  100%

Unpacking  [------------------------------------]    0%
Unpacking  [###---------------------------------]    8%
Unpacking  [######------------------------------]   16%
Unpacking  [#########---------------------------]   25%
Unpacking  [############------------------------]   33%
Unpacking  [###############---------------------]   41%
Unpacking  [##################------------------]   50%
Unpacking  [#####################---------------]   58%
Unpacking  [########################------------]   66%
Unpacking  [###########################---------]   75%
Unpacking  [##############################------]   83%
Unpacking  [#################################---]   91%
Unpacking  [####################################]  100%
Library Manager: ArduinoQueue@1.2.3 has been installed!
Dependency Graph
|-- WiFi @ 1.0
|-- ESPmDNS @ 1.0
|-- Update @ 1.0
|-- ArduinoQueue @ 1.2.3
Compiling /data/linp_doorbell/.pioenvs/linp_doorbell/src/esphome/components/api/api_connection.cpp.o
[... 97 lines cut ...]
Compiling /data/linp_doorbell/.pioenvs/linp_doorbell/FrameworkArduino/esp32-hal-timer.c.o
Compiling /data/linp_doorbell/.pioenvs/linp_doorbell/FrameworkArduino/esp32-hal-touch.c.o
Compiling /data/linp_doorbell/.pioenvs/linp_doorbell/FrameworkArduino/esp32-hal-uart.c.o
Compiling /data/linp_doorbell/.pioenvs/linp_doorbell/FrameworkArduino/libb64/cdecode.c.o
Compiling /data/linp_doorbell/.pioenvs/linp_doorbell/FrameworkArduino/libb64/cencode.c.o
Compiling /data/linp_doorbell/.pioenvs/linp_doorbell/FrameworkArduino/main.cpp.o
Compiling /data/linp_doorbell/.pioenvs/linp_doorbell/FrameworkArduino/stdlib_noniso.c.o
Compiling /data/linp_doorbell/.pioenvs/linp_doorbell/FrameworkArduino/wiring_pulse.c.o
Compiling /data/linp_doorbell/.pioenvs/linp_doorbell/FrameworkArduino/wiring_shift.c.o
Archiving /data/linp_doorbell/.pioenvs/linp_doorbell/libFrameworkArduino.a
Linking /data/linp_doorbell/.pioenvs/linp_doorbell/firmware.elf
RAM:   [=         ]  11.7% (used 38276 bytes from 327680 bytes)
Flash: [====      ]  44.6% (used 818030 bytes from 1835008 bytes)
Building /data/linp_doorbell/.pioenvs/linp_doorbell/firmware.bin
esp32_create_combined_bin(["/data/linp_doorbell/.pioenvs/linp_doorbell/firmware.bin"], ["/data/linp_doorbell/.pioenvs/linp_doorbell/firmware.elf"])
[... cut ...]
========================= [SUCCESS] Took 99.96 seconds =========================

********************************************************************************
We found 49.19MB of unnecessary PlatformIO system data (temporary files, unnecessary packages, etc.).
Use `pio system prune --dry-run` to list them or `pio system prune` to save disk space.
INFO Successfully compiled program.
INFO Connecting to 192.168.3.6
ERROR Connecting to 192.168.3.6:3232 failed: [Errno 111] Connection refused
```

ScorpionSPB, 2022-11-08:
Soldering iron + blow dryer + wromer32 = everything worked out. Thanks for the help.

pauln, 2024-11-06:
@BiznessDronss134 I'm not sure why you're putting files into `config/custom_components` - if that `config` folder is your Home Assistant config folder, that's definitely not right (ESPHome and Home Assistant are separate things).  If you're sure you have the `g03` variant of this doorbell (which is quite old by this point), you can try the [external_components config](https://github.com/pauln/esphome-linp-doorbell-g03/blob/feature/external_components/linp_doorbell.yaml) - which will automatically pull in the code for you when you run the ESPHome build.  If you've actually got the `g04` version, there's [a separate repository](https://github.com/pauln/esphome-linp-doorbell-g04) for that - which also uses `external_components`.

I haven't tried building the`g03` code for a long time, though, so it may not work - for one thing, you'll probably need to remove the `platformio_options` section (under `esphome:`) from the config, and replace the `esp32` section with the one from [the `g04` repo](https://github.com/pauln/esphome-linp-doorbell-g04/blob/main/doorbell.yaml).  Even that might not be enough, though - the `g03` code uses the ArduinoQueue library, which may not work on esp-idf.  If you've definitely got a `g03` doorbell, and can't get it to build using these notes (and the `external_components` branch linked above), please open a new issue for assistance - replying to old, closed issues is not the best way to get assistance (and may annoy other people who participated in the original issue thread).