pauln/esphome-linp-doorbell-g03 #23: Don't work on g6l-WIFI

000al000, 2021-12-02:
I 've got linptech doorbell with name G6l-wifi-rw

I flash firmware with config 

```
esphome:
  name: doorbell
  platform: ESP32
  board: esp32doit-devkit-v1
.
  platformio_options:
    platform: espressif32@1.11.0
    platform_packages: |-4
......
          framework-arduinoespressif32 @ https://github.com/pauln/arduino-esp32.git#solo-no-mac-crc/1.0.4

external_components:
  - source: github://pauln/esphome-linp-doorbell-g03@feature/external_components
    components: [ linp_doorbell ]

wifi:
  ssid: !secret
  password: !secret

logger:

api:
  password: !secret
ota:
  password: !secret

sensor:
  - platform: linp_doorbell
    volume:
      name: "Doorbell volume"
    chime_playing:
      name: "Doorbell playing tune"
#    use_old_service_names: True

```

and try call service from haas play_tune - no success (no sound).  Push button - no reaction.

```
[08:55:41][I][app:106]: ESPHome version 2021.8.0 compiled on Dec  2 2021, 08:45:05
[08:55:41][C][wifi:499]: WiFi:
[08:55:41][C][wifi:359]:   SSID: 'PASTE$COPY'
[08:55:41][C][wifi:360]:   IP Address: 10.100.10.94
[08:55:41][C][wifi:362]:   BSSID: 34:CE:00:65:30:79
[08:55:41][C][wifi:363]:   Hostname: 'doorbell'
[08:55:41][C][wifi:367]:   Signal strength: -53 dB ▂▄▆█
[08:55:41][C][wifi:371]:   Channel: 1
[08:55:41][C][wifi:372]:   Subnet: 255.255.255.0
[08:55:41][C][wifi:373]:   Gateway: 10.100.10.1
[08:55:41][C][wifi:374]:   DNS1: 10.100.10.1
[08:55:41][C][wifi:375]:   DNS2: 8.8.8.8
[08:55:41][C][linp_doorbell:041]: Doorbell Config:
[08:55:41][C][linp_doorbell:043]:   Volume 'Doorbell volume'
[08:55:41][C][linp_doorbell:043]:     State Class: ''
[08:55:41][C][linp_doorbell:043]:     Unit of Measurement: ''
[08:55:41][C][linp_doorbell:043]:     Accuracy Decimals: 0
[08:55:41][C][linp_doorbell:043]:     Icon: 'mdi:volume-high'
[08:55:41][C][linp_doorbell:044]:   Chime Playing 'Doorbell playing tune'
[08:55:41][C][linp_doorbell:044]:     State Class: ''
[08:55:41][C][linp_doorbell:044]:     Unit of Measurement: ''
[08:55:41][C][linp_doorbell:044]:     Accuracy Decimals: 0
[08:55:41][C][linp_doorbell:044]:     Icon: 'mdi:music'
[08:55:41][C][logger:189]: Logger:
[08:55:41][C][logger:190]:   Level: DEBUG
[08:55:41][C][logger:191]:   Log Baud Rate: 115200
[08:55:41][C][logger:192]:   Hardware UART: UART0
[08:55:41][C][ota:029]: Over-The-Air Updates:
[08:55:41][C][ota:030]:   Address: doorbell.local:3232
[08:55:41][C][ota:032]:   Using Password.
[08:55:41][C][api:095]: API Server:
[08:55:41][C][api:096]:   Address: doorbell.local:6053
[08:55:43][D][api.connection:730]: Client 'Home Assistant 2021.11.4 (10.100.10.200)' connected successfully!
[08:55:56][I][linp_doorbell:195]: Playing tune: 4
```

000al000, 2021-12-02:
[image]
[image]

000al000, 2021-12-02:
As i understant chip is CMT2217B. And it use SPI to communicate to ESP...
connecting rx2 and tx2 show no result.
So, now i got ESP with 220v power, but no doorbell :)

pauln, 2021-12-03:
Hmm, that's annoying.  My units also have the model number G6L-WIFI-RW printed on the case.  If you peel the "G6L-WIFI" sticker off the board (beside the ESP32), does it have a model/revision number printed on there?  Mine has "G6_MI v1.7", along with what appears to be a date code saying it was made in the 23rd week of 2019 ("D: 23 19").

I might order a new unit and see if I get one like yours, as I'm interested to see whether there are any differences in functionality which might make the new version more (or less) desirable... if you don't mind sharing, where did you order yours from?  (Mine came from Ali Express, but it was long enough ago that the store and/or listing are probably gone by now.)

000al000, 2021-12-03:
My version under label 
G6L_ESP32_B   (after B something like small superscipt "v" or may by it defect :) )
V 1.0
D: 3021 

Bought from Ali
https://aliexpress.ru/item/1005002571074600.html

My board power by 3,3 v (no 5v) and seems use spi to communicate

pauln, 2021-12-03:
Thanks - I've ordered one, but it'll probably take at least a few weeks to be delivered.

If anyone with this new revision is able to sniff the SPI communications on the original firmware, please share the details here; I'll get stuck into it when mine arrives (assuming it is the new revision), but if we can figure out the SPI comms before that, I'll try to work on it even without the hardware.

pauln, 2021-12-03:
For the record, the CMT2217B is a standalone OOK RF receiver, so unless the chip marked U2 (I can't see any identifying marks on the chip itself in your photo) is a microcontroller and/or doorbell chip, the ESP32 might be doing everything itself on this new version.  If that's the case, a much more comprehensive custom component will be needed to make the new version useful, but it hopefully also means that we'll be able to customise it more (and it won't have some of the weird quirks of the old version).

pauln, 2021-12-03:
> My version under label G6L_ESP32_B (after B something like small superscipt "v" or may by it defect :) ) V 1.0 D: 3021

I'm guessing that's G6L_ESP32_BY, as per the photo from @GenoMXXX in #22.

000al000, 2021-12-03:
Seems i've got ligtly different board that  @GenoMXXX
there is no "busy" pin point on my board, but pins for rx/tx are same.
[image]

pauln, 2021-12-14:
My order has arrived in record time, and at first glance, the chip marked U2 is indeed a doorbell chip (SZY8039B) similar to the FR0396-E2 on the original version.  This means that custom chimes are probably still off the table, but also confirms that the ESP32 is now responsible for receiving the button presses from the CMT2217B directly and telling the doorbell chip when to play which chime.

My current thinking for an ESPHome component for this new version is that I'll ditch the service calls for learning/forgetting buttons, and instead make the buttons and their respective chimes configurable via YAML.  When you press a button that's not configured, a debug log entry with the button's ID will be emitted, so you can easily identify buttons in order to add them to your config.  This should provide a good compromise between development complexity and flexibility; I can't make any promises, but hopefully I can put an initial version together reasonably soon.

000al000, 2021-12-14:
Thank you. I'll ready to be beta tester :)

nemoyar, 2022-01-03:
A new version has arrived, the old firmware does not work (( @pauln tell pls, are you working on a new version?

pauln, 2022-01-04:
Yes, I'm working on a new version.  It might look a bit different, as the new model doesn't have anything that's specific to it (the previous version had those connected to a secondary microcontroller which has its own serial protocol which it used to communicate with the ESP32) - just an RF receiver and a doorbell chime chip, plus some buttons and LEDs - so I'm currently leaning towards submitting PRs to ESPHome to support the doorbell buttons as a remote protocol and the doorbell chime chip as an output device, which would remove the need for a custom component entirely.  I'd then provide some config examples so that it's still easy to get up and running.

I've got the buttons working as a remote protocol, so once I've had a chance to build a component for the doorbell chime chip, I'll see how it all looks at that point to see whether I think it's worth bundling it all up into a custom component or whether landing support for the separate bits in core and providing example configs is indeed the way forward.

vampywiz17, 2022-01-08:
Hello guys! Somebody bought tuya version this doorbell?

If yes, it possible, please test it:

https://github.com/vampywiz17/tuyaiot2mqtt

pauln, 2022-01-08:
I wasn't aware that there's a Tuya version, but a quick Google suggests that there is indeed a Linptech G6L-TUYA version which looks pretty much identical from the outside.  That said, it's somewhat off-topic here since this project is about using ESPHome on the G6L-WIFI (and this issue is about the new variant of that model).

vampywiz17, 2022-01-08:
@pauln sorry the off topic!

pauln, 2022-01-09:
I've got the basics of the chime chip control working now, and have figured out a way to register remote protocols from a custom component - so I'll be able to put together a component which can be used without waiting for the protocols to be merged into ESPHome and a new version to be released.  I've got a bit more finishing off to do on the protocols, and then I can start putting together something that's ready to be shared.

nagyrobi, 2022-01-09:
> I've got the basics of the chime chip control working now

For the new version?

pauln, 2022-01-20:
Since the two versions are incompatible and the ESPHome components work completely differently, I've decided to split the new one off into a separate repository.  I noticed that the stock firmware on the new version refers to itself as `linp-doorbell-g04`, so I've gone with the same repository naming convention as this one: [pauln/esphome-linp-doorbell-g04](https://github.com/pauln/esphome-linp-doorbell-g04).  Please feel free to open issues and/or pull requests on the new repository if you come across any problems with the instructions or sample YAML configuration (or the protocols themselves, for that matter).

closed 2022-01-20 by pauln

plantroon, 2022-02-28:
> Hello guys! Somebody bought tuya version this doorbell?
> 
> If yes, it possible, please test it:
> 
> https://github.com/vampywiz17/tuyaiot2mqtt

Sorry to hijack the thread but have you bought it by any chance? I am wondering if it would work with Tuya-convert...