pauln/esphome-linp-doorbell-g03 #27: Workaround for broken ESP chip (using an another ESP with RF 433 Mhz receiver as replacement)

pcmester, 2022-03-14:
Hi there! Unfortunately, I broke the ESP chip on the board while soldering the UART connections for an OTA capable firmware update, and now the ESP pins are grounded and broken, no sign of life from it. I think the doorbell chip still works, but there isn't any understandable serial communication from that either, just the blue light flash while turning on.

My idea is that I can use a seperate ESP32 or even an 8266 with an RF 433 Mhz receiver to catch the doorbell button's signal, and then, use it as a trigger in Home Assistant for an automation. Is that possible? Could you please help me with that? (for example, what RF 433Mhz receiver can I use, and how should I configure ESPHome to catch the button's signal).

Thank you very much!

pauln, 2022-03-15:
There's a new version of the doorbell base unit which interfaces the ESP32 directly with the RF receiver and doorbell chime chip, which I've written an ESPHome remote protocol decoder for - [see my g04 repository](https://github.com/pauln/esphome-linp-doorbell-g04) for that (and a protocol for talking to the doorbell chime chip, though I'm not sure if it's compatible with the chime chip in the g03 model).

If you have a 433MHz OOK RF receiver which works similarly to the CMT2217B (outputting the demodulated signal on a pin), you should be able to configure a remote receiver with the `linptech_g6l` protocol from my g04 repo to detect the button presses.  Since I've developed the protocol decoder on the g04 hardware, it's probably best to at least start with an ESP32 (which has a built-in remote decoder peripheral which the ESP8266 doesn't have) - but if that works, you could definitely try it on an ESP8266.  I'd be interested to know how that goes, as I've been thinking about building a minimal ESPHome device to act as a general RF receiver (as I don't really want the original doorbell chimes - I'd rather have custom notifications via Home Assistant - so I don't need the original base station).

pcmester, 2022-03-15:
That's a great idea! I have tried something simliar but completly different before the ESP firmware came out. I have used the doorbell chip's serial communication via USB to serial to integrate it with HA, and then built some automations to use external notifications. 

Unfortunately, I don't have an RF receiver, so I would like to buy one. I am not an RF expert, so it is quite difficult. I have found this one: https://www.ebay.com/itm/284123740553
Maybe this is right for the job?

pauln, 2022-03-17:
I don't think that one would work with the ESPHome remote receiver, as it seems like a more fully featured transceiver which is intended to be configured to do the decoding as well as the demodulation, and give you nice serial data packets.  It's almost certainly capable of receiving the doorbell button transmissions, but you'd need to figure out how to configure it to do so.

I think that the "SYN480R" ASK/OOK receiver modules which can be found on ebay for a couple of dollars each, or something else along those lines, would would with the ESPHome remote receiver and the protocol decoder in my g04 repository - noting that you might need to buy or make a suitable antenna unless you can find a module which comes with one.

pcmester, 2022-04-20:
Thank you! Sorry for the late response. I have ordered one and I will try it with that.

pcmester, 2022-05-26:
Hi! I managed to get one and tested it several times but got no usable results. The received data is not coming into the log, not on wifi, not via serial. If I set the receiver to catch everything (with the "all" string), so not just the linp protocol messages, then I get something, but it is messy and feeds the log even when the receiver module is disconnected, so probalby just some glitch. I have also tried several ESP32 pins, but no success. What could went wrong?

pcmester, 2022-05-29:
Is it possible, that the old doorbell uses an another protocol for RF communication, therefore the g04's protocol decoder is not able to capture it?

pauln, 2022-05-30:
The old doorbell uses the same buttons - I've been using them interchangeably - so the RF protocol is the same.  It might be that the module isn't detecting the transmissions clearly, in which case replacing the antenna on the receiver module might help (I hear that a 17.3cm long solid core wire works well for 433MHz).  You could also try adjusting the `filter` and/or `idle` times (and maybe the `tolerance`) in the `remote_receiver` configuration, but I don't think you'll be able to change them much while staying within the limits of the protocol.  I guess the ideal option would be to find a CMT2210LB module, since we know that it's able to receive/decode the transmissions (since it's the chip used by the doorbell) - but a quick search hasn't revealed any available (other than a wholesale listing on Alibaba).

If tweaking the configuration and/or replacing the antenna doesn't help, your simplest option might be to buy a new doorbell and then use the appropriate ESPHome component for whichever version you receive; your existing button(s) will work with either version.

pauln, 2022-05-31:
@pcmester Was the RF receiver you got a SYN480R?  I just realised that I have a couple of those which I bought some time ago and never got around to using, so I might try it out with one of those in the next couple of days and see whether I can get it to work.

pcmester, 2022-07-04:
Hi! I still couldn't get it to work. Any update on your side?

pauln, 2022-07-04:
Sorry, my laptop died and I haven't quite got everything set up on the new one yet to get back into this.  I did have a quick go at it on the old laptop before it went up in smoke, but I didn't have much luck either - I might have to crack out my oscilloscope to get a proper look at the SYN480R's output and see if it's actually able to cope with the short duration of the doorbell's pulses.

pcmester, 2022-07-05:
Oh, that’s very unfortunate! Thank you!

spupetic, 2025-10-01:
Although I was not able to get the buttons and this component working with the Tuya board (despite updates from libretiny to fix the interrupt issues) or the SYN480R, RXB6 and RXB12 receivers, I managed to find an alternative solution that works for me on an ESP32, with one of the mentioned receivers:
```
globals:
  - id: last_trigger
    type: int
    restore_value: no
    initial_value: '0'
remote_receiver:
    pin:
      number: GPIO26
    tolerance: 50%
    filter: 20us         
    idle: 900us     
    clock_resolution: 500000 
    buffer_size: 16kb
    rmt_symbols: 512 
    on_raw:
      then:
        - lambda: |-
                int short_pulse_count = 0;
                int long_pulse_count = 0;

                for (int i = 0; i < x.size(); i++) {
                  int duration = abs(x[i]);

                 if (duration >= 50 && duration <= 150) {
                  short_pulse_count++;
                } else if (duration >= 210 && duration <= 550) {
                  long_pulse_count++;
                }
                   }
                // Condition: at least 10 short pulse and 3 long pulse
                if (short_pulse_count >= 10 && long_pulse_count >= 3) {
                  auto now = millis();
		// Limiting the trigger to 1 button press every 3 seconds
                  if (now - id(last_trigger) > 3000) {
                    ESP_LOGD("doorbell", "Doorbell button press detected!");
                    id(trigger_doorbell_event).execute();
                    id(last_trigger) = now;
                  }
                }
script:
  - id: trigger_doorbell_event
    then:
      - homeassistant.event:
          event: esphome.doorbell_pressed
          data:
            source: rf_remote
```

The idea is that you dump raw pulses and analyze them, what makes them differ from other signals or the noise, and try to match them in a lambda. My configuration might not work for you, you can add `dump: raw` to see your doorbell presses in the logs. Viewing the logs via USB instead of Wi-Fi is recommended.