pauln/esphome-linp-doorbell-g03 #28: This config is broken

vampywiz17, 2022-04-19:
Hello there,

This fw is broken since last ESPhome update. See this:

https://github.com/esphome/issues/issues/3176

Please fix it, it possible.

bonuzzz, 2022-04-25:
replace config.schema block in https://github.com/pauln/esphome-linp-doorbell-g03/blob/feature/external_components/components/linp_doorbell/sensor.py#L16
to:
```
CONFIG_SCHEMA = (
    cv.Schema(
        {
            cv.GenerateID(): cv.declare_id(LinpDoorbellComponent),
            cv.Optional(CONF_VOLUME): sensor.sensor_schema(
                unit_of_measurement=UNIT_EMPTY,
                accuracy_decimals=0,
            ),
            cv.Optional(CONF_CHIME_PLAYING): sensor.sensor_schema(
                unit_of_measurement=UNIT_EMPTY,
                accuracy_decimals=0,
            ),
            cv.Optional(CONF_USE_OLD_SERVICE_NAMES, default=False): cv.boolean,
        }
    )
)
```

vampywiz17, 2022-04-25:
@bonuzzz 

It fix it thanks!!!

closed 2022-04-27 by vampywiz17