pauln/esphome-linp-doorbell-g03 #29: Reboot every OTA firmware update

vampywiz17, 2022-12-01:
Hello there,

If  i update ESPhome, every time at need to try about 10-15 times that finally upload it. It restart when i upload the firmware via OTA.

Normal usage it not restart, only when i update it.

I can see it in the log:

```
Linking /data/csengo/.pioenvs/csengo/firmware.elf
RAM:   [=         ]  11.9% (used 38944 bytes from 327680 bytes)
Flash: [=====     ]  53.7% (used 986054 bytes from 1835008 bytes)
Building /data/csengo/.pioenvs/csengo/firmware.bin
esp32_create_combined_bin(["/data/csengo/.pioenvs/csengo/firmware.bin"], ["/data/csengo/.pioenvs/csengo/firmware.elf"])
Wrote 0x100c40 bytes to file /data/csengo/.pioenvs/csengo/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 96.03 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of csengo.local
INFO  -> 192.168.31.154
INFO Uploading /data/csengo/.pioenvs/csengo/firmware.bin (986176 bytes)
Uploading: [====================                                        ] 34% 
ERROR Error sending data: [Errno 104] Connection reset by peer

[20:07:58][C][mdns:104]:   Hostname: csengo
[20:07:58][C][ota:093]: Over-The-Air Updates:
[20:07:58][C][ota:094]:   Address: csengo.local:3232
[20:07:58][C][ota:097]:   Using Password.
[20:07:58][W][ota:103]: Last Boot was an unhandled reset, will proceed to safe mode in 7 restarts
[20:07:58][C][api:138]: API Server:
[20:07:58][C][api:139]:   Address: csengo.local:6053
[20:07:58][C][api:141]:   Using noise encryption: YES
```

nagyrobi, 2022-12-01:
Typical case of weak WiFi coverage, or interference (close to other network equipment/psu) in the place where you have placed the ESP.

vampywiz17, 2022-12-01:
@nagyrobi 

First i think the same. But i try a lot of places and it do the same everywhere... Other ESP-s with same place update without any problem.

nagyrobi, 2022-12-01:
Bad ESP...

plantroon, 2022-12-03:
This may sound stupid but try rebooting your wifi router/ap. Do you run OpenWRT on it by any chance? I have this same problem and that's what helped.

vampywiz17, 2022-12-03:
I use Unifi AP-s. it restarted a many times.

plantroon, 2022-12-03:
> I use Unifi AP-s. it restarted a many times.
Try restarting everything and flashing as soon as possible after the restart. Every time I change configuration of any esphome device I have to do that before flashing. But it may be a different problem entirely of course...

Btw, I see your nick @vampywiz17 in a lot of places regarding doorbells and sonoff rf bridge as well - did you by any chance make ANY kinetic button work with sonoff rf bridge or any other similar device? This setup from pauln works perfectly for me, but I'd still like to know if Sonoff bridge can be made work with these kinetic buttons.

vampywiz17, 2022-12-07:
@plantroon 

Yes, i able to use a Kerui self powered doorbell with rtl433.