pauln/esphome-linp-doorbell-g03 #31: where are all button sensors on external_component branch?

vecjh, 2023-03-29:
Hi, I can no longer compile this on the mail branch. But it went OK with external_component branch. However, it doesn't have button sensors? How can they be added?

pauln, 2023-03-30:
The external_components branch doesn't have the button sensors, instead focusing on events.  You should be able to change the trigger in your HA automations to be an Event trigger, using `esphome.linp_doorbell_button_pressed`, and set the event data to `button: "1"` (replacing `1` with the appropriate button number); if you have more than one of these doorbell units, you can also add `device: "your_device_name"` to listen to button presses from a specific unit.  You can use the Events tab on the Developer Tools page in HA to listen for these events in order to check which button number and device name to use.

vecjh, 2023-03-30:
@pauln thanks a lot for prompt help! Much appreciated. That worked indeed. I was about to find it myself through digging the code. But was unsure of the HA's naming convention of events. Documentation is poor on 3rd party event naming. So I'm closing the issue.

closed 2023-03-30 by vecjh