pauln/esphome-linp-doorbell-g03 #34: An error occurred during installation. error: ArduinoQueue.h

BiznessDronss134, 2024-11-06:
Good afternoon, there was an error compiling the file
`INFO ESPHome 2024.10.0
INFO Reading configuration /config/esphome/linp-doorbell-g03.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing linp-doorbell-g03 (board: esp32doit-devkit-v1; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Library Manager: Installing ArduinoQueue @ 1.2.3
Warning! Could not find the package with 'ArduinoQueue @ 1.2.3' requirements for your system 'linux_x86_64'
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.6
Compiling .pioenvs/linp-doorbell-g03/src/esphome/components/linp_doorbell/linp_doorbell.cpp.o
Compiling .pioenvs/linp-doorbell-g03/src/esphome/components/logger/logger_esp8266.cpp.o
Compiling .pioenvs/linp-doorbell-g03/src/esphome/components/logger/logger_host.cpp.o
Compiling .pioenvs/linp-doorbell-g03/src/esphome/components/logger/logger_libretiny.cpp.o
Compiling .pioenvs/linp-doorbell-g03/src/esphome/components/logger/logger_rp2040.cpp.o
Compiling .pioenvs/linp-doorbell-g03/src/esphome/components/md5/md5.cpp.o
In file included from src/esphome/components/linp_doorbell/linp_doorbell.cpp:1:
src/esphome/components/linp_doorbell/linp_doorbell.h:6:10: fatal error: ArduinoQueue.h: No such file or directory

**********************************************************************
* Looking for ArduinoQueue.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ArduinoQueue.h"
* Web  > https://registry.platformio.org/search?q=header:ArduinoQueue.h
*
**********************************************************************

 #include "ArduinoQueue.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/linp-doorbell-g03/src/esphome/components/linp_doorbell/linp_doorbell.cpp.o] Error 1
========================== [FAILED] Took 3.75 seconds ==========================
`

My configuration

`esphome:
  name: linp-doorbell-g03
  friendly_name: linp-doorbell-g03

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: arduino
          
external_components:
  - source: github://pauln/esphome-linp-doorbell-g03@feature/external_components
    components: [ linp_doorbell ]

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "Bhm2966+69J+d/vOr9ePcHbN0BK9U6oVfYql92n3kpI="

ota:
  - platform: esphome
    password: "248464279afd8e6b637286d716d080c2"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  
  manual_ip:
    # Установите для этого IP-адрес ESP
    static_ip: 192.168.1.34
    # Установите для этого значение IP-адреса маршрутизатора. Часто заканчивается на .1
    gateway: 192.168.1.1
    # Подсеть сети. 255.255.255.0 работает для большинства домашних сетей.
    subnet: 255.255.255.0  
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Linp-Doorbell-G03"
    password: "HZjl90s0o9g6"

captive_portal:
sensor:
  - platform: linp_doorbell
    volume:
      name: "Doorbell volume"
    chime_playing:
      name: "Doorbell playing tune"
#    use_old_service_names: True`

BiznessDronss134, 2024-11-06:
[pauln](https://github.com/pauln)  can you help?

nemo0823, 2025-01-26:
@BiznessDronss134 @pauln  
I was able to compile the firmware. I used ESPHome version 2022.12.3
To fix the error, I used version 1.2.5 of the ArduinoQueue library to fix the line in the `sensor.py` file 
   ` cg.add_library(“ArduinoQueue”, “1.2.3”)` to ` cg.add_library(“ArduinoQueue”, “1.2.5”)`

```
esphome:
  name: linp_doorbell
  libraries:
    - "ArduinoQueue"          
  platformio_options:
    platform: espressif32@1.11.2
    platform_packages:
      - framework-arduinoespressif32 @ https://github.com/pauln/arduino-esp32.git#solo-no-mac-crc/1.0.6

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: arduino

external_components:
  - source: github://pauln/esphome-linp-doorbell-g03@feature/external_components
    components: [ linp_doorbell ]

wifi:
  ssid: "USER"
  password: "PASS"
logger:
# Enable Home Assistant API
api:
  #password: 'api'

# Enable over-the-air updates
ota:
  #password: !secret ota_password

sensor:
  - platform: linp_doorbell
    volume:
      name: "Doorbell volume"
    chime_playing:
      name: "Doorbell playing tune"
#    use_old_service_names: True```

BiznessDronss134, 2025-02-28:
> [@BiznessDronss134](https://github.com/BiznessDronss134) [@ pauln](https://github.com/pauln)Мне удалось скомпилировать прошивку. Я использовал ESPHome версии 2022.12.3Чтобы исправить ошибку, я использовал версию 1.2.5 библиотеки ArduinoQueue, чтобы исправить строку в `sensor.py` ` cg.add_library(“ArduinoQueue”, “1.2.3”)`с ` cg.add_library(“ArduinoQueue”, “1.2.5”)`
> 
> ```
> esphome:
>   name: linp_doorbell
>   libraries:
>     - "ArduinoQueue"          
>   platformio_options:
>     platform: espressif32@1.11.2
>     platform_packages:
>       - framework-arduinoespressif32 @ https://github.com/pauln/arduino-esp32.git#solo-no-mac-crc/1.0.6
> 
> esp32:
>   board: esp32doit-devkit-v1
>   framework:
>     type: arduino
> 
> external_components:
>   - source: github://pauln/esphome-linp-doorbell-g03@feature/external_components
>     components: [ linp_doorbell ]
> 
> wifi:
>   ssid: "USER"
>   password: "PASS"
> logger:
> # Enable Home Assistant API
> api:
>   #password: 'api'
> 
> # Enable over-the-air updates
> ota:
>   #password: !secret ota_password
> 
> sensor:
>   - platform: linp_doorbell
>     volume:
>       name: "Doorbell volume"
>     chime_playing:
>       name: "Doorbell playing tune"
> #    use_old_service_names: True```
> ```

Вот как у меня получилось 

```
esphome:
  name: linp_doorbell
  libraries:
    - "ArduinoQueue"          
  platformio_options:
    platform: espressif32@1.11.2
    lib_deps:
      - ArduinoQueue@1.2.5
    platform_packages:
      - framework-arduinoespressif32 @ https://github.com/pauln/arduino-esp32.git#solo-no-mac-crc/1.0.6

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: arduino

external_components:
  - source: github://pauln/esphome-linp-doorbell-g03@feature/external_components
    components: [ linp_doorbell ]
```